platformio/platform-atmelavr #293: avrdude.conf No such file or directory

Thijxx, 2022-10-31:


ivankravets, 2022-10-31:
Please check examples here https://docs.platformio.org/en/latest/platforms/atmelavr.html#upload-using-programmer

Thijxx, 2022-11-02:
> Please check examples here https://docs.platformio.org/en/latest/platforms/atmelavr.html#upload-using-programmer

You replied with a URL that I mentioned in my ticket as being the source of the problem.

robmaxtech, 2022-11-17:
The example in the doc works fine; you should specify the avrdude.conf:

upload_flags =
    -C
    ; use "tool-avrdude-megaavr" for the atmelmegaavr platform
    ${platformio.packages_dir}/tool-avrdude/avrdude.conf

and the programmer

    -c
    <your programmer>

HTH!

ivankravets, 2022-11-18:
@robmaxtech, thanks for the confirmation.

@Thijxx, please reopen this issue if you still have any problems with PIO Core.

closed 2022-11-18 by ivankravets