platformio/platform-atmelavr #297: Setting fuses on Attiny fails due to spaces in path to avrdude

robmaxtech, 2022-11-17:
Hello,

I encountered an issue with the atmelavr platform - I could do the upload, but setting fuses failed - the path to avrdude.exe contained spaces and therefore it was not found, I found a similar issue here, regarding upload: [#10](https://github.com/platformio/platform-atmelavr/issues/10  )
Looking at the change, I modified the corresponding location in fuses.py and it seems to work ~~removed part~~ strikethrough:

FUSESUPLOADERFLAGS=[
    "-p",
    "$BOARD_MCU",
    "-C",
    ~~'"%s"'~~
    ~~%~~ join(env.PioPlatform().get_package_dir("tool-avrdude") or "", "avrdude.conf"),

Is this correct? Maybe it can be included officially then.

bongo505, 2022-11-23:
Same thing worked for me. I knew that main.py worked and copied from there but ended up with the same as you did

tactical-snacks, 2023-03-09:
Had the same issue with setting the fuses on an ATmega328PB with an atmel-ice. The fuses.py edit above worked.

The error I received when running Set Fuses ...
```
$ pio run -t fuses -v
Processing ATmega328PB (platform: atmelavr; board: ATmega328PB; framework: arduino; board_build.f_cpu: 16000000L; upload_protocol: atmelice_isp; upload_port: usb; board_fuses.hfuse: 0xDC; board_fuses.lfuse: 0xDE; board_fuses.efuse: 0xFD)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/ATmega328PB.html
PLATFORM: Atmel AVR (4.1.0) > ATmega328PB
HARDWARE: ATMEGA328PB 16MHz, 2KB RAM, 32KB Flash
PACKAGES:
 - framework-arduino-avr-minicore @ 2.2.1
 - tool-avrdude @ 1.70100.0 (7.1.0)
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 8 compatible libraries
Scanning dependencies...
Building in release mode

TARGET CONFIGURATION:
---------------------
Target = atmega328pb
Clock speed = 16000000L
Oscillator = external
BOD level = 2.7v
Save EEPROM = yes
UART port = uart0
Clock output = no
CFD enable = no
---------------------

Selected fuses: [lfuse = 0xDE, hfuse = 0xDC, efuse = 0xFD]
avrdude -p atmega328pb -C ""C:\Users\first last\.platformio\packages\tool-avrdude\avrdude.conf"" -e -c atmelice_isp -Ulock:w:0x0f:m -Uhfuse:w:0xDC:m -Ulfuse:w:0xDE:m -Uefuse:w:0xFD:m
avrdude OS error: cannot open config file C:\Users\first: No such file or directory
avrdude error: unable to process system wide configuration file C:\Users\first
*** [fuses] Error 1
```
If i set the fuses in the upload_flags, I didn't have an issue. Re-installed vscode and pio and a lot of other muckin around before I saw this.

MCUdude, 2023-05-09:
I'm just wondering, does the provided solution work if _only_ the single quotes are removed?
I currently have PR #311 open where I've already done modifications to fuses.py and bootloader.py, so I might as well add this fix to the PR as well...

thijses, 2023-09-26:
this issue appears to have popped back up again.
scons env.Replace() is adding quotes around any text with spaces if it's a list of strings, so for example:
```
env.Replace(
    FUSESUPLOADERFLAGS=[
        "-C",
        '"%s"'
        % join(env.PioPlatform().get_package_dir("tool-avrdude") or "", "avrdude.conf"),
    ]
)
```
produces:
> -C ""path_with_spaces.avrdude.conf""

instead of:
> -C "path_with_spaces.avrdude.conf"

this (presumably intended to be helpful) quirk of env.Replace() causes issues in fuses.py and bootloader.py.
I have submitted [a PR with a patch/fix](https://github.com/platformio/platform-atmelavr/pull/319)

P.S. i was pointed towards this issue by @maxgerhardt over [on the PIO forums](https://community.platformio.org/t/avrdude-config-file-not-found-because-of-spaces-in-path/35391/7).

commit 90aa1f1d (2023-09-28): Excessive quotes fix for AVRDUDE flags for bootloader and fuses (#319)

Resolves #297
closed 2023-09-28 by valeros