platformio/platform-atmelavr #301: Update all MCUdude Arduino cores to the latest version

MCUdude, 2023-01-18:
Hi!

I've now released new versions of all my Arduino cores, and it would be great if you could add them + the new Avrdude 7.1 toolchain to PlatformIO.

The following releases _requires_ Avrdude 7.1:

[MightyCore v2.2.0](https://github.com/MCUdude/MightyCore/releases)
[MiniCore v2.2.0](https://github.com/MCUdude/MiniCore/releases)
[MajorCore v2.2.0](https://github.com/MCUdude/MajorCore/releases)
[MegaCore v2.2.1](https://github.com/MCUdude/MegaCore/releases)
[MicroCore v2.2.1](https://github.com/MCUdude/MicroCore/releases)

MegaCore also adds a few new AVR targets, so please merge #286 and #287 while you're at it.

MegaCoreX also needs to be updated ([v1.1.1](https://github.com/MCUdude/MegaCoreX/releases), requires Avrdude 7.1), but this needs to be done in the [platform-atmelmegaavr](https://github.com/platformio/platform-atmelmegaavr) repo.

The new release requires Avrdude 7.1. The Arduino developers have released a version with lots of binaries you guys can use directly:
https://github.com/arduino/avrdude-packing/releases/tag/7.1-arduino.1

Avrdude 7.1 includes lots of new features and improvements, and it also does extensive error checking in the avrdude.conf files 3rd party cores bundles. I wouldn't recommend using Avrdude 7.1 on old Arduino cores that haven't updated their avrdude.conf file. But for all MCUdude Arduino cores, Avrdude 7.1 is required.

Also, there's no need to provide a separate Avrdude version _just_ for the `patform-atmelmegaavr` repo. It can absolutely be combined with the `platform-atmelavr` one.

Thanks for the support @valeros and @ivankravets

To wrap it up:
|                                                                                      | Avrdude 6.3 | Avrdude 7.1 |
|-----------------------------------------------------------|--------------|----------------|
| All other Arduino cores not updated for Avrdude 7.1 | ✅               | ❌                |
| MightyCore v2.1.3 and below                                      | ✅               | ❌                |
| MiniCore v2.1.3 and below                                          | ✅               | ❌                |
| MajorCore v2.1.3 and below                                        | ✅               | ❌                |
| MegaCore **v2.2.0** and below                                  | ✅               | ❌                |
| MicroCore **v2.2.0** and below                                  | ✅               | ❌                |
| MegaCoreX v1.1.0 and below                                      | ✅               | ❌                |
| MightyCore v2.2.0 and above                                      | ❌               | ✅                |
| MiniCore v2.2.0 and above                                          | ❌               | ✅                |
| MajorCore v2.2.0 and above                                        | ❌               | ✅                |
| MegaCore **v2.2.1** and above                                  | ❌               | ✅                |
| MicroCore **v2.2.1** and above                                  | ❌               | ✅                |
| MegaCoreX v1.1.1 and above                                      | ❌               | ✅                |

valeros, 2023-01-20:
Hi @MCUdude ! Thanks for such a comprehensive description. All your latest Arduino cores and the AVRDUDE 7.1 are published to the PlatformIO registry.

closed 2023-01-20 by valeros

MCUdude, 2023-01-20:
Thank you @valeros!

I have a large PlatformIO project where I'm using the ATmega1284P, and after running `pio pkg update`, it's using MightyCore v2.1.3 instead of 2.2.0. Is there anything special I need to do in order to get the latest version?

Oh! And could you please also update platform-atmelmegaavr with the latest MegaCoreX version and Avrdude? 🙂 

``` ini
; Part of the platformio.ini file

; Common settings for all targets
[env]
platform = atmelavr
framework = arduino
; Chip target
board = ATmega1284P
; Clock frequency in [Hz]
board_build.f_cpu = 8000000L
```

valeros, 2023-01-20:
A new version of the `atmelavr` dev-platform hasn't been released yet, so if you want to test it out, you will need to install the upstream version of the platform:
```ini
; Common settings for all targets
[env]
platform = https://github.com/platformio/platform-atmelavr.git
framework = arduino
...
```

The `atmelmegaavr` platform will be updated as well a bit later.

MCUdude, 2023-01-20:
@valeros Now that I realized that PlatformIO uses the avrdude.conf file that's bundled with the Avrdude binary and not the one that's shipped with the Arduino core, It _should_ be safe to use Avrdude 7.1 with all AVR targets. It doesn't introduce any breaking changes as long as the avrdude.conf file is updated to work with Avrdude 7.1. Bear this in mind as the 7.1 release is a big improvement over 6.3.

commit 11df7eb9 (2023-01-20): Update MCUdude Arduino cores

Resolve #301