platformio/platform-atmelavr #314: framework-arduino-avr package incredibly outdated

Yannik, 2023-05-25:
The latest release for the [framework-arduino-avr](https://registry.platformio.org/tools/platformio/framework-arduino-avr) package is from 3 years ago. 
The [source repository](https://github.com/arduino/ArduinoCore-avr) has had several releases since then. It would be great to get this package updated. 

Since the package is owned by platformio, I hope opening an issue here is correct.

fermino, 2023-09-25:
Is there anything we can do to help this move forward? There are some features missing that are a deal breaker for some projects!
Thanks! :)

fermino, 2023-09-25:
For people in need of using the latest version of the Arduino Core (1.8.6 as of today), I will be maintaing a repository with an upstream copy of the core with minimal modifications to make it work with platformio (package.json). It isn't my idea to fork this package, or to be maintaining this forever, but to have a working package until this one can be updated. For that very same reason I will not be publishing that repo in platformio's registry. If you need a different version let me know so I can add it :)

To use this repository in your project: 
```
platform_packages =
	framework-arduino-avr @ https://github.com/fermino/framework-arduino-avr-upstream#upstream-1.8.6
```

labodj, 2023-10-08:
@fermino 
first of all thanks for your effort, I'm compiling in Linux and there is a problem using your repo

config:
```ini
[env]
platform = https://github.com/platformio/platform-atmelavr.git
platform_packages = https://github.com/fermino/framework-arduino-avr-upstream#upstream-1.8.6
framework = arduino
board = controllino_maxi
```

error:
```
Resolving release dependencies...
Tool Manager: Installing platformio/framework-arduino-avr @ ~5.1.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: framework-arduino-avr@5.1.0 has been installed!
Tool Manager: Installing framework-arduino-avr-upstream @ *
UnknownPackageError: Could not find the package with 'framework-arduino-avr-upstream @ *' requirements for your system 'linux_x86_64'
```

fermino, 2023-10-08:
> [env]platform = https://github.com/platformio/platform-atmelavr.gitplatform_packages = https://github.com/fermino/framework-arduino-avr-upstream#upstream-1.8.6framework = arduinoboard = controllino_maxi
> Tool Manager: ***@***.*** has been installed!

Hey :)
Try changing platform to "atmelavr".

labodj, 2023-10-08:
Just tested it... no luck

`UnknownPackageError: Could not find the package with 'framework-arduino-avr-upstream @ *' requirements for your system 'linux_x86_64'`

fermino, 2023-10-08:
@labodj I didn't see it, you're missing the `framework-arduino-avr @` part in platform_packages. Without that PIO assumes the package name is framework-arduino-avr-upstream, which has the extra `-upstream`

Try with this: 
```
platform = atmelavr
platform_packages = framework-arduino-avr @ https://github.com/fermino/framework-arduino-avr-upstream#upstream-1.8.6
framework = arduino
board = controllino_maxi
```

labodj, 2023-10-09:
@fermino thanks it was an oversigth, now it works as expected!
Anyway it's compatible with few boards (EG: not controllino)...
I just opened a pull request to add current platformio supported boards here:  https://github.com/fermino/framework-arduino-avr-upstream/pull/1

commit cd16c257 (2023-11-24): Update Arduino AVR core to v1.8.6

Resolves #314
closed 2023-11-24 by valeros