platformio/platform-atmelavr #320: Support for Urboot and Mighty/Mini/Mega/MajorCore v3.0.0

MCUdude, 2023-10-25:
Version 3.0.0 of Mighty/Mini/Mega/MajorCore has been released, and Optiboot has been replaced with Urboot.
Urboot is superior to Optiboot in virtually every way (Read more about the excellent Urboot bootloader [here](https://github.com/stefanrueger/Urboot)), but it will require different fuse settings. Optiboot is still bundled, but Urboot should be the default bootloader.

| Status | Task |  Can be done without further ado | Who?|
|-----|------|-------|-----|
| ❌ | [Update Avrdude to 7.2-arduino.1](https://github.com/arduino/avrdude-packing/releases/tag/7.2-arduino.1)<br/>(7.1 is backwards compatible with 7.1, so all cores<br/> that uses 7.1 can be upgraded to use 7.2) | **YES** | @valeros or @ivankravets  |
| ❌ | Update Mighty/Mini/Mega/MajorCore to v3.0.0 | **YES** | @valeros or @ivankravets |
| ❌ | Update fuses and bootloader script to support </br>Urboot, and make Urboot the default bootloader<br/> for all Mighty/Mini/Mega/MajorCore targets | **NO**, a PR for this are needed first | @MCUdude |

When everything is done, I'm thinking about providing a template like this to help users get the most out of these cores (the current version of the template[ can be found here](https://github.com/MCUdude/MightyCore/blob/8b653d5ad0d0e55247eb5c8df618ffe142afde04/PlatformIO.md)):

```ini
[platformio]
default_envs = Upload_UART ; Default build target

; Common settings for all environments
[env]
platform = atmelavr
framework = arduino

; TARGET SETTINGS
board = ATmega1284P
board_build.f_cpu = 16000000L
board_build.variant = standard

; BUILD OPTIONS
build_unflags =
build_flags = 

; SERIAL MONITOR OPTIONS
monitor_port = ${env:Upload_UART.upload_port}
monitor_speed = 9600

; Run the following command to upload with this environment
; pio run -e Upload_UART -t upload
[env:Upload_UART]
upload_protocol = urclock
upload_port = /dev/cu.usbserial*
; Urboot has auto-baud detection, so "any" baud rate can be used
board_upload.speed = 250000

; Run the following command to upload with this environment
; pio run -e Upload_ISP -t upload
[env:Upload_ISP]
upload_protocol = custom
upload_flags =
  -C$PROJECT_PACKAGES_DIR/tool-avrdude/avrdude.conf
  -p$BOARD_MCU
  -cusbasp
upload_command = avrdude $UPLOAD_FLAGS -U flash:w:$SOURCE:i

; Run the following command to set fuses
; pio run -e fuses_bootloader -t fuses
; Run the following command to set fuses + burn bootloader
; pio run -e fuses_bootloader -t bootloader
[env:fuses_bootloader]
board_hardware.oscillator = external ; Oscillator type
board_hardware.uart = uart0          ; Set UART to use for serial upload (For Optiboot, use "uart0_optiboot")
board_hardware.bod = 2.7v            ; Set brown-out detection
board_hardware.eesave = yes          ; Preserve EEPROM when uploading using programmer
upload_protocol = usbasp             ; Use the USBasp as programmer
upload_flags =                       ; Select USB as upload port and divide the SPI clock by 8
  -PUSB
  -B8
```

#### Bottom line:
* `upload_protocol = urclock` are backwards compatible with `upload_protocol = arduino`
* Millions(!) of pre-compiled Urboot bootloader binaries can be found [here](https://github.com/stefanrueger/Urboot.hex), including the ones bundled with Mighty/Mini/Mega/MajorCore
* Optiboot will still be available, but `board_hardware.uart = uartN_optiboot` has to be specified to burn Optiboot. `upload_protocol = arduino` is also required for uploading using the Optiboot bootloader

valeros, 2023-10-25:
Hi @MCUdude, the packages have been uploaded:
- https://registry.platformio.org/tools/platformio/tool-avrdude/versions
- https://registry.platformio.org/tools/platformio/framework-arduino-avr-minicore/versions
- https://registry.platformio.org/tools/platformio/framework-arduino-avr-megacore/versions
- https://registry.platformio.org/tools/platformio/framework-arduino-avr-mightycore/versions
- https://registry.platformio.org/tools/platformio/framework-arduino-avr-majorcore/versions

MCUdude, 2023-10-26:
Thanks @valeros!
I'll try to find some time next week to implement Urboot support for the fuses.py and bootloader.py scripts.

commit b7abdb5b (2023-11-28): Add support for the Urboot bootloader (#321)

Resolves #320
closed 2023-11-28 by valeros