platformio/platform-atmelavr #322: error: 'fabs' was not declared in this scope

dustinlacewell, 2023-11-05:
When I try to compile a simple hello world with platformio, I get compilation errors within the platform toolkit:

```cpp
#include "Arduino.h"

void setup()
{
  
}

void loop()
{
  
}
```

```
Processing uno (platform: https://github.com/platformio/platform-atmelavr.git; board: uno; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/uno.html
PLATFORM: Atmel AVR (4.2.0+sha.265921d) > Arduino Uno
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 31.50KB Flash
DEBUG: Current (avr-stub) External (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr @ 5.1.0
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 8 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Servo @ 1.2.1
|-- NewPing @ 1.9.7
|-- IRremote @ 4.2.0
Building in release mode
Compiling .pio\build\uno\src\main.cpp.o
Compiling .pio\build\uno\lib004\Servo\avr\Servo.cpp.o
Compiling .pio\build\uno\lib004\Servo\mbed\Servo.cpp.o
Compiling .pio\build\uno\lib004\Servo\megaavr\Servo.cpp.o
Compiling .pio\build\uno\lib004\Servo\nrf52\Servo.cpp.o
Compiling .pio\build\uno\lib004\Servo\renesas\Servo.cpp.o
Compiling .pio\build\uno\lib004\Servo\sam\Servo.cpp.o
Compiling .pio\build\uno\lib004\Servo\samd\Servo.cpp.o
Compiling .pio\build\uno\lib004\Servo\stm32f4\Servo.cpp.o
Compiling .pio\build\uno\liba52\NewPing\NewPing.cpp.o
Archiving .pio\build\uno\libFrameworkArduinoVariant.a
Compiling .pio\build\uno\FrameworkArduino\CDC.cpp.o
Compiling .pio\build\uno\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\uno\FrameworkArduino\HardwareSerial0.cpp.o
In file included from C:\Users\Dustin\.platformio\packages\framework-arduino-avr\cores\arduino/USBAPI.h:27:0,
                 from C:\Users\Dustin\.platformio\packages\framework-arduino-avr\cores\arduino/Arduino.h:234,
                 from src\main.cpp:1:
c:\users\dustin\.platformio\packages\toolchain-atmelavr\avr\include\util\delay.h: In function 'void _delay_ms(double)':
c:\users\dustin\.platformio\packages\toolchain-atmelavr\avr\include\util\delay.h:184:32: error: 'fabs' was not declared in this scope
   __ticks_dc = (uint32_t)(ceil(fabs(__tmp)));
                                ^~~~
c:\users\dustin\.platformio\packages\toolchain-atmelavr\avr\include\util\delay.h:184:32: note: suggested alternative: 'labs'       
   __ticks_dc = (uint32_t)(ceil(fabs(__tmp)));
                                ^~~~
                                labs
Compiling .pio\build\uno\FrameworkArduino\HardwareSerial1.cpp.o
c:\users\dustin\.platformio\packages\toolchain-atmelavr\avr\include\util\delay.h:184:27: error: 'ceil' was not declared in this scope
   __ticks_dc = (uint32_t)(ceil(fabs(__tmp)));
                           ^~~~
c:\users\dustin\.platformio\packages\toolchain-atmelavr\avr\include\util\delay.h:184:27: note: suggested alternative: 'cli'        
   __ticks_dc = (uint32_t)(ceil(fabs(__tmp)));
                           ^~~~
                           cli
Compiling .pio\build\uno\FrameworkArduino\HardwareSerial2.cpp.o
c:\users\dustin\.platformio\packages\toolchain-atmelavr\avr\include\util\delay.h: In function 'void _delay_us(double)':
c:\users\dustin\.platformio\packages\toolchain-atmelavr\avr\include\util\delay.h:273:32: error: 'fabs' was not declared in this scope
   __ticks_dc = (uint32_t)(ceil(fabs(__tmp)));
                                ^~~~
c:\users\dustin\.platformio\packages\toolchain-atmelavr\avr\include\util\delay.h:273:32: note: suggested alternative: 'labs'       
   __ticks_dc = (uint32_t)(ceil(fabs(__tmp)));
                                ^~~~
                                labs
c:\users\dustin\.platformio\packages\toolchain-atmelavr\avr\include\util\delay.h:273:27: error: 'ceil' was not declared in this scope
   __ticks_dc = (uint32_t)(ceil(fabs(__tmp)));
                           ^~~~
c:\users\dustin\.platformio\packages\toolchain-atmelavr\avr\include\util\delay.h:273:27: note: suggested alternative: 'cli'        
   __ticks_dc = (uint32_t)(ceil(fabs(__tmp)));
                           ^~~~
                           cli
*** [.pio\build\uno\src\main.cpp.o] Error 1
```

valeros, 2023-11-06:
Hi @dustinlacewell, might be a broken toolchain package for some reason. Could you delete the `c:\users\dustin\.platformio\packages` folder and try again?

dustinlacewell, 2023-11-07:
It was because one of my own files was named `math.h`. Changing it to `maths.h` resolved the issue.

closed 2023-11-07 by dustinlacewell