platformio/platform-atmelavr #323: Add support for ATX2

QuadraBoy, 2023-11-27:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 11

**PlatformIO Version**: 3.4.4 (Home), 6.1.11 (Core)

### Description of problem

ATX2 is not included in the supported boards list. It would be lovely if Platform IO supports it so I, and many more people, don't need to download the necessary software to program an ATX2.

MCUdude, 2024-01-11:
You didn't provide any information about the board, whatsoever. I've never heard of an ATX2 board until now.
But after a bit of research, it seems like it's based on an ATmega644 of some sort.

The ATmega644 _is_ supported by PlatformIO, so all that's needed is a tailored platformio.ini file that matches the ATX2 hardware setup. No need for a dedicated board implementation. Use this as a base: https://github.com/MCUdude/MightyCore/blob/master/PlatformIO.md

QuadraBoy, 2024-01-11:
Sorry for not providing any information because I thought ATX2 was a very common board. Thanks for the information!