platformio/platform-atmelavr #324: toolchain-atmelavr source code and build scripts?

mid-kid, 2023-11-30:
Hi, I'm trying to figure out a few details about the toolchain-atmelavr package:
* Which specific version of the code is used? Is it patched?
* How is it built? With what kind of options?
* Where are the current binaries obtained from, if anywhere?

valeros, 2023-12-01:
Hi @mid-kid, 

> Where are the current binaries obtained from, if anywhere?

We use binaries provided by the Arduino IDE, for example: https://github.com/arduino/Arduino/blob/master/hardware/package_index_bundled.json#L74-L116

> How is it built? With what kind of options?

You can find the used option if you run `avr-gcc -v`, for example:

```
Target: avr
Configured with: ../gcc/configure --enable-fixed-point --enable-languages=c,c++ --prefix=/home/jenkins-mingw32/workspace/avr-gcc-staging/label/Ubuntu14.04x64-mingw32/objdir --disable-nls --disable-libssp --disable-libada --disable-shared --with-avrlibc=yes --with-dwarf2 --disable-doc --target=avr --host=i686-w64-mingw32
```

> Which specific version of the code is used? Is it patched?

It's GCC 7.3.0 with some additional patches, you can find sources here https://github.com/arduino/toolchain-avr

valeros, 2023-12-04:
Package definitions are specified in this file https://github.com/platformio/platform-atmelavr/blob/develop/platform.json, the packages are hosted on the [PlatformIO Package Registry](https://registry.platformio.org/) and delivered through our CDN network.

ivankravets, 2023-12-05:
PlatformIO Core handles all these things automatically. When you build a project for the first time, PlatformIO Core pulls dependencies from the PlatformIO Registry and validates checksum for each downloaded package. See https://github.com/platformio/platformio-core/blob/develop/platformio/package/download.py#L122

mid-kid, 2023-12-09:
Yeah, I'm just trying to understand the "repository" structure and how it decides to download what from where. Mostly out of personal interest, but also because I'm wary of package managers downloading random binaries from the internet.

ivankravets, 2023-12-09:
PlatformIO Core is an OSS project. You learn more about PlatformIO Package Manager from its sources https://github.com/platformio/platformio-core/tree/develop/platformio/package/manager

The core logic is based on [get_systype()](https://github.com/platformio/platformio-core/blob/77e6d1b09956cf38fa8d2722bd9c88919f5c43c2/platformio/util.py#L138). Having information about host architecture, PlatormIO Core goes to the PlatformIO Registry and looks for the best package corresponding to the https://semver.org/

mid-kid, 2023-12-10:
I see, it's querying https://api.registry.platformio.org/v3/packages/platformio/tool/toolchain-atmelavr for all the information, including download URL (from dl.registry.platformio.org) and checksum.

Is the data obtained from api.registry.platformio.org hosted in a git somewhere, or is it entered into a live database? Is the backend published anywhere?

thirstyice, 2024-05-14:
Hi, I'm also having trouble understanding how this works.
Where is the source for the toolchain-atmelavr package? I haven't been able to find it anywhere, even though the license is listed as GPL-2.0-or-later

valeros, 2024-05-14:
@thirstyice 

> Where is the source for the toolchain-atmelavr package? I haven't been able to find it anywhere, even though the license is listed as GPL-2.0-or-later

It's generic GCC 7.3.0 with some additional patches, you can find sources here https://github.com/arduino/toolchain-avr

thirstyice, 2024-05-15:
How are they packaged for platformio?

valeros, 2024-05-15:
> How are they packaged for platformio?

We use compiled packages provided by Arduino IDE from [their manifest](https://downloads.arduino.cc/packages/package_index.json).