platformio/platform-atmelavr #327: please consider to add upload.mcu

onikcode, 2023-12-29:
Please consider to add upload.mcu
like existing build.mcu
in case upload.mcu absent then pio would use build.mcu
https://github.com/platformio/platform-atmelavr/blob/d9d2738bcb1c632e2f5e3c57318e6ae8281ee960/builder/main.py#L207C20-L207C20

it needs to support Amperka boards

they compiles code as build.mcu=atmega328p
but upload as upload.mcu=tmega328pb

see https://github.com/amperka/amperka-boards/blob/master/platform.txt#L11C8-L11C8

MCUdude, 2024-01-11:
If you're uploading to an ATmega328PB, why not compile for it as well? PlatformIO supports the ATmega328PB.

And if you just _have_ to upload to a different target than what you have compiled for, you can always use a custom upload command:

From the [MiniCore PlatformIO README](https://github.com/MCUdude/MiniCore/blob/master/PlatformIO.md):
```ini
; Run the following command to upload with this environment
; pio run -e Upload_ISP -t upload
[env:Upload_ISP]
; Custom upload procedure
upload_protocol = custom
; Avrdude upload flags
upload_flags =
  -C$PROJECT_PACKAGES_DIR/tool-avrdude/avrdude.conf
  -p$BOARD_MCU
  -PUSB
  -cusbasp
; Avrdude upload command
upload_command = avrdude $UPLOAD_FLAGS -U flash:w:$SOURCE:i
```

onikcode, 2024-01-11:
>why not compile for it as well
not everything compile well :( - yes, this looks as issue to amperka but i thought about agile of platform io.
upload_command  - I use this currently.