platformio/platform-atmelavr #330: Add a board config to 'Board Explorer' for standalone 328P PU (DIP28 IC on breadboard) with 8MHz and 1MHz internal clock

crunchysteve, 2024-07-06:
'Board Explorer' needs a standalone config for 328P PU (DIP28 IC on breadboard) with 8MHz and 1MHz internal clock.

I've tried to generate one from the docs but that low-level stuff just breaks my brain as well as breaking a few ICs in my attempts. Even pros would benefit from having such a setup in the Board Explorer. Being able to copy paste the default setup is a productivity aid. I have a project I can't progress in PlatformIO and will need to regress to ArduinoIDE without a naked chip board config for the 328P PU.

Cheers,
Crunchy,
Melbourne Australia

DjordjeMandic, 2024-12-27:
This breaks the point of existence of minicore. I think that examples for platformio.ini for this would be more useful than custom board definition.

crunchysteve, 2024-12-27:
Fair enough. I'll defer to the experts, because I'm not an expert by any stretch and an example that works would definitely help. (As opposed to my own failed attempts to make a standalone 328 project.)

I'm happy for the maintainers to close this issue.

closed 2024-12-27 by crunchysteve