platformio/platform-atmelavr #333: Warning about maximum data size should be an error instead

robbieh, 2024-09-06:
```
Checking size .pio/build/32u416m/firmware.elf                                                     
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"                        
RAM:   [========Warning! The data size (7243 bytes) is greater than maximum allowed (2560 bytes)                                                                                                    
==]  282.9% (used 7243 bytes from 2560 bytes)                                                                                                                                                       
Flash: [===       ]  25.9% (used 7432 bytes from 28672 bytes)                                     
Building .pio/build/32u416m/firmware.hex                                                                                                                                                            
Configuring upload protocol...                                                                                                                                                                      
AVAILABLE: avr109                                                                                                                                                                                   
CURRENT: upload_protocol = avr109                                                                 
Looking for upload port...                                                                        
Auto-detected: /dev/ttyACM0                                                                                                                                                                         
Forcing reset using 1200bps open/close on port /dev/ttyACM0                                                                                                                                         
Waiting for the new upload port...                                                                                                                                                                  
Uploading .pio/build/32u416m/firmware.hex                           
```

This destroyed the board I was working with. This should have failed with an error instead.

valeros, 2024-10-10:
Hi @robbieh 

Indeed, PlatformIO only prints a warning in case of a RAM overflow, but it shouldn't break your board. Could you please elaborate on what happened after you uploaded the binary?

robbieh, 2024-10-10:
The board stopped responding to attempts to program it over USB. I don't have a non-USB programmer handy, and I suspect using one would fix it.