platformio/platform-atmelavr #341: Upload to Arduino Nano (CH341 USB chip) fails in PlatformIO but works in Arduino IDE

nilton61, 2025-04-21:
Environment Details

PlatformIO Core: 6.1.18
PlatformIO Home: 3.4.4
OS: Windows 10 Pro (version 2009, build 19045.5737)
Board: Arduino Nano (with CH341 USB-to-serial chip)
Hardware: Intel Xeon E3-1505M v5 @ 2.80GHz, 16GB RAM

Description
I'm experiencing consistent upload failures to Arduino Nano boards when using PlatformIO. The same code uploads successfully through Arduino IDE without any issues.
The problem persists across:

Multiple different Arduino Nano boards
Different computers
Fresh PlatformIO projects
Minimal code examples

Error Messages
When attempting to upload, I receive the following error:

valeros, 2025-04-22:
Hi @nilton61, the error message is missing.

nilton61, 2025-04-22:
Sorry, i forgot. 
I changed to this:
board = nanoatmega328new
that resolved the issue

closed 2025-04-23 by valeros