platformio/platform-atmelavr #342: platform atmel avr v5.1.0 failing to upload

jozles7777, 2025-05-13:
I got a new pc windows11 

no way to upload on my homemade atmega328 boards via usb and cp2102 or ch340
I got various errors depending on configuration of upload protocol and speed
after hours I finally tried with the old release of  platform and the old platformio.ini file 
and it works !

working platformio.ini (atmelavr@3.3.0)

[env:ATmega328P]
;platform = atmelavr
platform = platformio/atmelavr@3.3.0

board = ATmega328P
board_hardware.oscillator = internal
board_build.f_cpu = 8000000L
framework = arduino
monitor_port = COM4
monitor_speed = 115200
upload_port = COM4
;upload_protocol = urclock
;upload_protocol = avrisp2
;upload_protocol = avr911
;-B 125kHz
;-xstrict
; -xdelay=10000
upload_speed = 38400
;115200
;38400

PS : DTR signal which makes reset to ATmega is different depending of pio release

valeros, 2025-05-14:
Hi @jozles7777, what's the error exactly?

jozles7777, 2025-05-14:
Hi !

with that in platformio.ini (v5.1.0)

[env:ATmega328P]
platform = atmelavr

board = ATmega328P
board_hardware.oscillator = internal
board_build.f_cpu = 8000000L
framework = arduino
monitor_port = COM4
monitor_speed = 115200
upload_port = COM4
;upload_protocol = urclock
;upload_protocol = avrisp2
;upload_protocol = avr911
;-B 125kHz
;-xstrict
; -xdelay=10000
upload_speed = 38400
;115200
;38400

I get that

Configuring upload protocol...
AVAILABLE: urclock
CURRENT: upload_protocol = urclock
Looking for upload port...
Using manually specified: COM4
Uploading .pio\build\ATmega328P\firmware.hex
avrdude: AVR device initialized and ready to accept instructions
avrdude error: bootloader might be optiboot 8.0? Please use -xbootsize=<num>

avrdude error: unable to read signature data for part ATmega328P, rc=-1
avrdude error: unable to read signature data, rc=-1

avrdude done.  Thank you.

*** [upload] Error 1

jozles7777, 2025-05-14:
with that :

[env:ATmega328P]
platform = atmelavr

board = ATmega328P
board_hardware.oscillator = internal
board_build.f_cpu = 8000000L
framework = arduino
monitor_port = COM4
monitor_speed = 115200
upload_port = COM4
;upload_protocol = urclock
;upload_protocol = avrisp2
upload_protocol = avr911
;-B 125kHz
;-xstrict
; -xdelay=10000
upload_speed = 38400
;115200
;38400

I get that 

Configuring upload protocol...
AVAILABLE: urclock
CURRENT: upload_protocol = urclock
Looking for upload port...
Using manually specified: COM4
Uploading .pio\build\ATmega328P\firmware.hex
avrdude: AVR device initialized and ready to accept instructions
avrdude error: bootloader might be optiboot 8.0? Please use -xbootsize=<num>

avrdude error: unable to read signature data for part ATmega328P, rc=-1
avrdude error: unable to read signature data, rc=-1

avrdude done.  Thank you.

*** [upload] Error 1

valeros, 2025-05-14:
Please also share here the logs of a successful upload using v3.3.0

jozles7777, 2025-05-14:
I had that too with different other config ....

Configuring upload protocol...
AVAILABLE: avrisp
CURRENT: upload_protocol = avrisp
Looking for upload port...
Using manually specified: COM4
Uploading .pio\build\ATmega328P\firmware.hex
avrdude error: programmer is not responding
avrdude warning: attempt 1 of 10: not in sync: resp=0x00
avrdude warning: attempt 2 of 10: not in sync: resp=0x72
avrdude warning: attempt 3 of 10: not in sync: resp=0x61
avrdude warning: attempt 4 of 10: not in sync: resp=0x64
avrdude warning: attempt 5 of 10: not in sync: resp=0x20
avrdude warning: attempt 6 of 10: not in sync: resp=0x69
avrdude warning: attempt 7 of 10: not in sync: resp=0x6e
avrdude warning: attempt 8 of 10: not in sync: resp=0x69
avrdude warning: attempt 9 of 10: not in sync: resp=0x74
avrdude warning: attempt 10 of 10: not in sync: resp=0x20
avrdude error: unable to open programmer avrisp on port COM4

avrdude done.  Thank you.

*** [upload] Error 1

valeros, 2025-05-14:
I'm a bit lost here. v3.3.0 also throws an error: `avrdude error: unable to open programmer avrisp on port COM4`, so avrdude wasn't able to program the target.

Also, you have two different upload protocols `avr911` and `avrisp`. Which one is correct?

closed 2025-05-14 by valeros